arras-energy/gridlabd-old #1078: AWS endpoints no longer work

dchassin, 2022-01-12:
## Problem description

A number of important AWS endpoints return errors. Please verify that *all* endpoints in `gridlabd/S3` work correctly and fix S3/Route 53/CloudFront as needed.

## Steps to reproduce

Try the following commands:

1. `git clone https://source.gridlabd.us/`
2. `curl https://code.gridlabd.us/master/README.md`.

## Expected behavior

1. should clone the source code into `gridlabd` folder.
2. should download the master README file.

catzzz, 2022-01-12:
2. Fixed `curl https://code.gridlabd.us/master/README.md.`  error by changing the redirect url `github.com/slacgismo/gridlabd/raw/` in S3 to `raw.githubusercontent.com/slacgismo/gridlabd/`.

catzzz, 2022-01-24:
1. Fixed by resetting the AWS Cloudfront endpoint.

closed 2022-01-24 by catzzz