arras-energy/gridlabd-old #1083: Installation under arm environment does not work

catzzz, 2022-01-13:
## Problem description

The installation of Girdlabd in arm environment with docker image(centos:8) does not work

## Steps to reproduce
Under ARM environments such as Mac M1 and AWS ec2 arm environment with docker image(centos:8)
Create a temporary Dockerfile:
~~~
FROM centos:8
RUM yum install git -y
WORKDIR /usr/local/src
RUN git clone -b develop https://github.com/slacgismo/gridlabd.git
WORKDIR /usr/local/src/gridlabd
RUN ./install.sh -v -t 
~~~

## System information

* GridLAB-D version: 
* Operating system: centos:8 under Mac M1 

## Supporting information

## Attachments
~~~
ERROR: No matching distribution found for Shapely==1.7.1
~~~

catzzz, 2022-01-13:
Changing `Shapely>=1.7.1` in `gldcore/geodata/requirements.txt`and `gldcore/scripts/requirements` results other errors:
~~~
#37 142.6   ERROR: Command errored out with exit status 1:
#37 142.6    command: /usr/local/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nofjt9ug/fiona_78d858ff1e3d4f3da41e382373b1f476/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nofjt9ug/fiona_78d858ff1e3d4f3da41e382373b1f476/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-gabx6sxy
#37 142.6        cwd: /tmp/pip-install-nofjt9ug/fiona_78d858ff1e3d4f3da41e382373b1f476/
#37 142.6   Complete output (2 lines):
#37 142.6   Failed to get options via gdal-config: [Errno 2] No such file or directory: 'gdal-config'
#37 142.6   A GDAL API version must be specified. Provide a path to gdal-config using a GDAL_CONFIG environment variable or use a GDAL_VERSION environment variable.
#37 142.6   ----------------------------------------
#37 142.6 WARNING: Discarding https://files.pythonhosted.org/packages/ec/f7/093890341a7e8fbfcdfa04caf4dfb588ebab32c13ceaa6a3819da79ea106/Fiona-1.8.20.tar.gz#sha256=a70502d2857b82f749c09cb0dea3726787747933a2a1599b5ab787d74e3c143b (from https://pypi.org/simple/fiona/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
#37 143.1   ERROR: Command errored out with exit status 1:
#37 143.1    command: /usr/local/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nofjt9ug/fiona_63cd6256b6cc435c860eab73ba122a5b/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nofjt9ug/fiona_63cd6256b6cc435c860eab73ba122a5b/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-y6h7jm8p
#37 143.1        cwd: /tmp/pip-install-nofjt9ug/fiona_63cd6256b6cc435c860eab73ba122a5b/
#37 143.1   Complete output (2 lines):
#37 143.1   Failed to get options via gdal-config: [Errno 2] No such file or directory: 'gdal-config'
#37 143.1   A GDAL API version must be specified. Provide a path to gdal-config using a GDAL_CONFIG environment variable or use a GDAL_VERSION environment variable.
#37 143.1   ----------------------------------------
#37 143.1 WARNING: Discarding https://files.pythonhosted.org/packages/a0/d9/6042aeb073d11341f7726de0586ff71c13117c34959dcf07bd4ee6d4b93e/Fiona-1.8.19.tar.gz#sha256=b9059e0b29c2e9e6b817e53f941e77e1aca7075f986005d38db307067b60458f (from https://pypi.org/simple/fiona/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
#37 143.1 ERROR: Could not find a version that satisfies the requirement Fiona>=1.8.19 (from versions: 0.5, 0.6, 0.6.1, 0.6.2, 0.7, 0.8, 0.9, 0.9.1, 0.10, 0.12, 0.12.1, 0.13, 0.14, 0.15, 0.16, 0.16.1, 1.0, 1.0.1, 1.0.2, 1.0.3, 1.1, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.1.5, 1.1.6, 1.2.0, 1.3.0, 1.4.0, 1.4.1, 1.4.2, 1.4.3, 1.4.4, 1.4.5, 1.4.6, 1.4.7, 1.4.8, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.6.3, 1.6.3.post1, 1.6.4, 1.7b1, 1.7.0, 1.7.0.post2, 1.7.1, 1.7.1.post1, 1.7.2, 1.7.3, 1.7.4, 1.7.5, 1.7.6, 1.7.7, 1.7.8, 1.7.9, 1.7.9.post1, 1.7.10, 1.7.10.post1, 1.7.11, 1.7.11.post1, 1.7.11.post2, 1.7.12, 1.7.13, 1.8a1, 1.8a2, 1.8a3, 1.8b1, 1.8b2, 1.8rc1, 1.8.0, 1.8.1, 1.8.2, 1.8.3, 1.8.4, 1.8.5, 1.8.6, 1.8.7, 1.8.8, 1.8.9, 1.8.9.post1, 1.8.9.post2, 1.8.10, 1.8.11, 1.8.12, 1.8.13, 1.8.13.post1, 1.8.14, 1.8.15, 1.8.16, 1.8.17, 1.8.18, 1.8.19, 1.8.20)

catzzz, 2022-01-13:
Updating `gdal`and `fiona` installation results validation error:
~~~
#54 160.2 ERROR: Cannot install scipy==1.6.0 and scipy==1.6.2 because these package versions have conflicting dependencies.
~~~

catzzz, 2022-01-13:
update `scipy >= 1.6.2` in three `requirements.txt` files. It passes the `make system` command. Then it results validation error: 
~~~
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata.glm error unexpected, code 5 (simulation failed) in 1.0 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_address.glm error unexpected, code 5 (simulation failed) in 1.2 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_elevation.glm error unexpected, code 5 (simulation failed) in 1.2 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_utility.glm error unexpected, code 5 (simulation failed) in 1.1 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/geodata/autotest/test_datasets.glm error unexpected, code 5 (simulation failed) in 4.7 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_matrix_linalg.glm error unexpected, code 5 (simulation failed) in 11.1 seconds
~~~
The error output:
~~~
cat /usr/local/src/gridlabd/gldcore/scripts/autotest/test_matrix_linalg/gridlabd.err
ERROR [gridlabd-assert]: test '3+2.2128127e-08j,3-2.2128127e-08j;0.70710678+0j,0.70710678-0j;0.70710678+7.8234742e-09j,0.70710678-7.8234742e-09j; == 3+2.9802322e-08j,3-2.9802322e-08j;0.70710678+0j,0.70710678-0j;0.70710678+1.0536712e-08j,0.70710678-1.0536712e-08j;' failed!
ERROR    [INIT] : GldMain::subcommand(format='%s/gridlabd-%s',...): command '/usr/local/opt/gridlabd/4.2.30-220113-develop/bin/gridlabd-assert $(gridlabd matrix -f linalg.eig "1,2;-2,5" | tr '\n' ';') == '3+2.9802322e-08j,3-2.9802322e-08j;0.70710678+0j,0.70710678-0j;0.70710678+1.0536712e-08j,0.70710678-1.0536712e-08j;'
' returns code 1
[root@897787abc5a4 gridlabd]# cat /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata/gridlabd.err
Traceback (most recent call last):
  File "/usr/local/opt/gridlabd/4.2.30-220113-develop/bin/gridlabd-geodata", line 358, in <module>
    import geopandas
  File "/usr/local/lib/python3.9/site-packages/geopandas/__init__.py", line 7, in <module>
    from geopandas.io.file import _read_file as read_file  # noqa
  File "/usr/local/lib/python3.9/site-packages/geopandas/io/file.py", line 20, in <module>
    from fiona import Env as fiona_env
  File "/usr/local/lib/python3.9/site-packages/fiona/__init__.py", line 85, in <module>
    with fiona._loading.add_gdal_dll_directories():
AttributeError: partially initialized module 'fiona' has no attribute '_loading' (most likely due to a circular import)
ERROR    [INIT] : GldMain::subcommand(format='%s/gridlabd-%s',...): command '/usr/local/opt/gridlabd/4.2.30-220113-develop/bin/gridlabd-geodata create 37.4205,-122.2046
' returns code 1
~~~

dchassin, 2022-01-13:
Note that the requirement should only be in one place, i.e., at the high level in the code tree needed to ensure that all resources which use it have it.  Usually `./requirements.txt` is best.

catzzz, 2022-02-01:
Pass installation 
Fail in validation 
~~~~
Starting validation test in directory '/usr/local/src/gridlabd'
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata.glm error unexpected, code 5 (simulation failed) in 1.0 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_address.glm error unexpected, code 5 (simulation failed) in 1.2 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_elevation.glm error unexpected, code 5 (simulation failed) in 1.2 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata_utility.glm error unexpected, code 5 (simulation failed) in 1.1 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/geodata/autotest/test_datasets.glm error unexpected, code 5 (simulation failed) in 11.6 seconds
ERROR    [INIT] : /usr/local/src/gridlabd/gldcore/scripts/autotest/test_matrix_linalg.glm error unexpected, code 5 (simulation failed) in 13.0 seconds
Processing ./gridlabd/module/tape/autotest/test_tape_whitespace.glm...                                           
Validation report:
695 models tested
6 unexpected errors
689 tests succeeded
>99% success rate
~~~~
### Error output 
~~~
[root@cb152834c626 gridlabd]# cat /usr/local/src/gridlabd/gldcore/scripts/autotest/test_geodata/gridlabd.err
Traceback (most recent call last):
  File "/usr/local/opt/gridlabd/4.2.30-220120-develop/bin/gridlabd-geodata", line 358, in <module>
    import geopandas
  File "/usr/local/lib/python3.9/site-packages/geopandas/__init__.py", line 7, in <module>
    from geopandas.io.file import _read_file as read_file  # noqa
  File "/usr/local/lib/python3.9/site-packages/geopandas/io/file.py", line 20, in <module>
    from fiona import Env as fiona_env
  File "/usr/local/lib/python3.9/site-packages/fiona/__init__.py", line 85, in <module>
    with fiona._loading.add_gdal_dll_directories():
AttributeError: partially initialized module 'fiona' has no attribute '_loading' (most likely due to a circular import)
ERROR    [INIT] : GldMain::subcommand(format='%s/gridlabd-%s',...): command '/usr/local/opt/gridlabd/4.2.30-220120-develop/bin/gridlabd-geodata create 37.4205,-122.2046
' returns code 1
~~~

closed 2023-04-15 by dchassin