arras-energy/gridlabd-old #1087: Missing default timezone setup in the installation process under Windows WSL/Debian.

catzzz, 2022-01-19:
## Problem description

Missing default timezone setup in the installation process under Wondows 10 WSL/Debian:11

## Steps to reproduce
Under Windows 10  WSL/Debian:11 
Run `cat /etc/timezone` the output is `Etc/GMT+8`. It causes installation error. 
 

## System information

* GridLAB-D version: HiPAS GridLAB-D 4.2.30-220119
* Operating system: 
4.4.0-19041-Microsoft #1237-Microsoft Sat Sep 11 14:32:00 PST 2021 x86_64 GNU/Linux

dchassin, 2022-01-19:
We do not have a current PR for Debian 11.  Do we need on yet?

catzzz, 2022-01-20:
set Local timezone from web 
`ln -snf /usr/share/zoneinfo/$(curl https://ipapi.co/timezone) /etc/localtime`

closed 2022-01-20 by catzzz