arras-energy/gridlabd-old #1096: Extract XY coordinate data from CYME converter

jo-hal, 2022-02-04:
## Background

It’s difficult to plot feeder data on a map or canvas without coordinate data. CYME database contains coordinate data. This data would be useful for plotting a feeder in a map view.  This data does not need to be the form of latitude and longitude, just X and Y data directly extract directly from CYME is enough.

## Proposed Solution

Extract XY from CYME and save it as a separate CSV file? This is the most straightforward way without modification to GLD itself. The attachment “Glow_NodeCoordinates.csv” is an example of file name and format that we are currently using in GLOW.

node,x,y
load_1,-119.0022,35.38528
load_2,-119.0022,35.38577
node_3,-119.0022,35.3846

## Alternatives

Alternative 1, since this is not latitude and longitude, it may need to add new parameters in GLD object. For example, “coordinate_x”, “coordinate_y” parameters.

Alternative 2, extract XY coordinate data and use it as value in latitude and longitude.

## Example file "Glow_NodeCoordinates.csv"
[Glow_NodeCoordinates.csv](https://github.com/slacgismo/gridlabd/files/8006825/Glow_NodeCoordinates.csv)

dchassin, 2023-04-15:
@MitchellAV, can you confirm this has been fixed? If so, please close this issue.

MitchellAV, 2023-04-17:
@dchassin Need to confirm, will close issue if fixed