arras-energy/gridlabd-old #1197: Positive value limitation of “undervoltage_violation_threshold” property in Node class

yeyanzhu, 2022-08-05:
## Problem description
The "“undervoltage_violation_threshold” property in Node class is limited to positive value. Negative value will be reset to zero. 
The node under-voltage violation can not be detected.

## Steps to reproduce
The autotest model file and output csv file is attached
Using the attached model file to reproduce the issues.

## Expected behavior
Remove the positive value limitation on “undervoltage_violation_threshold” , so the node under-voltage violation can be detected.

## System information
* GridLAB-D version: 
HiPAS GridLAB-D 4.3.1-220710

## Attachments
[test_node_voltage_violation.zip](https://github.com/slacgismo/gridlabd/files/9270380/test_node_voltage_violation.zip)

dchassin, 2022-08-05:
We need to fix the spec because this means assigning the global default when under/over voltage is zero will also not work any longer.  For now, I'm going to change the implementation so that the default values are assigned at creation instead of initialization.

dchassin, 2022-08-05:
Now we will see a warning the `under > over`.  Otherwise, the values are accepted as given and will override the defaults `under=-default` and `over=+default`.

closed 2023-04-15 by dchassin