arras-energy/gridlabd-old #1268: (Provide a one-line synopsis of the problem)

MateusGheorghe, 2023-04-04:
## Problem description

Hi!
I am trying to download gridlabd following the developer quickstart.
I was able to clone the repository using 
`git clone https://github.com/slacgismo/gridlabd gridlabd`
However, when I try to install it later using 
`gridlabd/install.sh`
I get the following error: 

> build-aux/setup.sh
> ERROR: gridlabd/install.sh failed -- see /usr/local/var/gridlabd/install.log for details

There are no install.log generated so I cannot get any more details. 
I am trying to install gridlabd in a machine with Ubuntu 22.04.2

## Steps to reproduce
Use a machine with Ubuntu 22.04.2
1. Run `git clone https://github.com/slacgismo/gridlabd gridlabd`
2. Run `gridlabd/install.sh`
3. Error:
> build-aux/setup.sh
> ERROR: gridlabd/install.sh failed -- see /usr/local/var/gridlabd/install.log for details

## Expected behavior

I would expect gridlabd would be installed.

## System information

* GridLAB-D version: latest version from github
* Operating system: Ubuntu 22.04.2

dchassin, 2023-04-10:
This is being addressed by #1145 which is pending resolution of a number of open PRs on this very question.

dchassin, 2023-04-15:
@MateusGheorghe, in the interim, you can install from the `develop` branch instead of `master`, e.g.,

~~~
git clone https://source.gridlabd.us/ /usr/local/src/gridlabd -b develop --depth 1
cd /usr/local/src/gridlabd
./install.sh
Password: *****
autoreconf -isf
./configure
make system
gridlabd --validate
~~~

Alternatively, you can use one of the pre-built docker containers located at https://docker.gridlabd.us/.  We maintain these for users who can't or don't want/need to build gridlabd from source.

closed 2023-04-15 by dchassin