arras-energy/gridlabd-old #1284: Potential CYME 9 Rev 6 MDB to GLM conversion issue

MitchellAV, 2023-05-01:
## Problem description

When using the GridLAB-D CYME MDB to GLM converter, certain MDB files that are in the latest version of CYME (CYME 9.0 Rev 6) is unable to successfully convert into a working GLM file. This error seems to be inconsistent as several CYME 9 Rev 6 files were tested of different sizes and the issue was not present. Need to test further with other CYME MDB files that are from the latest version of CYME to be able to confirm that the current converter is working correctly or not.

## Steps to reproduce

1. Converting an MDB file to GLM using the GridLAB-D cli
e.g. `gridlabd -D mdb_load_options="-t cyme" <mdb_filename>.mdb`

## Expected behavior

Should be able to supply an MDB file to the GridLAB-D command line and be able to use the MDB to GLM converter with the CYME flag to converter the CYME file to GLM.

## System information

* GridLAB-D version: (Insert output of `gridlabd --version=all` here)
* Operating system: (Insert output of `uname -a` here)

## Supporting information

No further supporting information available at this time.

## Attachments

(If applicable, attach screenshots, supporting files, and other data need to help explain and reproduce the problem. If you encountered a program fault or internal error, and you have built GridLAB-D with `-O0 -g`, then on Linux systems you can attach the output of the following command:)
~~~
bash$ ulimit -c unlimited
bash$ gridlabd <your command options...>
... crash dumps core ...
bash$ gdb -batch -ex "backtrace" $(gridlabd --version=install) ./core
~~~

(If you are running on macOS, you must use `lldb` to do the following.)
~~~
bash$ ulimit -c unlimited
bash$ ls -lt /cores # find the most recent core dump
bash$ lldb -c /cores/core.<number>
(lldb) bt -- backtrace
~~~

dchassin, 2023-05-19:
@MitchellAV, is this still an open issue?

MitchellAV, 2023-05-20:
@dchassin Yes, this is still an open issue. Once we obtain the files I requested from SCE, we should be able to move forward with determining if this issue is valid or not.

PR #1299 (2023-06-26): Fixes #1284 - Updated CYME converter to solve outstanding problems
Fixes #1284 

Following changes to the CYME converter
- Fixed issue with certain MDB files not being able to be converted and erroring out
- Solved to infinite loop issue in voltage_checks by setting loop limit to much smaller number and informing the user within the GLM file and output that the converter was unable to resolve the voltage mismatch
- Set default line length to 1 m if line length is 0
- Added option `GLM_INPUT_LINE_LENGTH_UNIT` in config.csv to allow for user to define line length unit to be `m` or `km` so that l [... cut]
commit 4111e71f (2023-06-26): Fixes #1284 - Updated CYME converter to solve outstanding problems (#1299)

* Changes to CYME converter to fix issues with converting mdb files
* Updated documentation for mdb converter
closed 2023-07-09 by dchassin