arras-energy/gridlabd-old #1327: (Provide a one-line synopsis of your enhancement request)

Plavit, 2023-08-10:
## Background
I was installing GridLAB-D in the past few days for a research project and it proved unnecessarily complex, I propose a few changes to Readme that will solve this issue.

## Proposed solution
Three changes proposed to README.md for better clarity:

1. Remove the **Contributions** section, as https://source.gridlabd.us/blob/master/CONTRIBUTING.md does not exist.

2. Add the following to the **Windows WSL** section:
GridLAB-D supports several WSL distributions, including the latest Ubuntu versions. See the setup folder for currently supported distributions.

3. Add the following to the **Windows WSL** section below the validation code `./build.sh --system --validate` :
The validation may take some time and it need not complete completely successfully for GridLAB-D to be used. The final report may look something like this:

```
Validation report:
747 models tested
1 unexpected errors
746 tests succeeded
>99% success rate
Total validation elapsed time: 281.9 seconds
See '/home/username/gridlabd/validate.txt' for details
ERROR [build.sh]: unable to verify install 
```

## Alternatives
Feel free to amend my proposed wording

## Additional context
Not needed

dchassin, 2023-08-11:
@Plavit, have you considered using the pre-built images instead of building it yourself?

Plavit, 2023-08-11:
I built it fine in the end, however it's possible that we might need to make some source tweaks in our research project so I needed to build from source along with my colleagues.

I am therefore humbly proposing minor edits in the readme that, in my opinion, would help others that will use the WSL approach in the future. 

I do realise of course it is up to your consideration whether to include the suggestions, however I do believe that it would be a useful edit for those that will consider working with the tool. 🙏

dchassin, 2023-08-11:
@Plavit, thanks for the suggestions.